pierremolinaro/acan2517 #1: ESP32 compatibility

btdieselworks, 2018-11-27:
Hi Pierre- Great job with these new ACAN2515 and ACAN2517 drivers! Most other MCP2515/2517 drivers seem to be lacking in various areas, but your ACAN drivers look excellent and very easy to use. Im looking forward to experimenting with them.

One question I had was regarding Espressif ESP32 compatibility. The Arduino ESP32 SPI drivers are a bit different in that they do not support the spi.usingInterrupt function. When I compile the ACAN2515/2517 test programs for ESP32, the compiler gives an error because the "usingInterrupt" does not exist in ESP32.

Just as a quick experiment, I modified your source files (ACAN2515.cpp/ACAN2517.cpp) and commented out line 156 "mSPI.usingInterrupt (itPin)".

On my work bench here with an MCP2517 development board (MCP2517 with a 20Mhz Ceralock resonator), with that mSPI.usingInterrupt line removed, the ACAN2517 driver initializes "ok", and it does **send** out CAN messages correctly. It does NOT receive any messages correctly though.

My tests were done with low speed single-wire CAN (33.333k) used on some GM/Opel/Saab vehicles, with a TH8056 single-wire-CAN transceiver connected to the MCP2517.

I could not make it send or receive with traditional 500k CAN. Perhaps something wrong with oscillator settings or my use of a 20MHz resonator and not a true 40MHz crystal?

I assume the receiving problem was due to the ACAN driver not correctly using the interrupt trigger from the MCP2517 due to the fact that I removed the usingInterrupt function.

Do you have any ideas on how to modify the driver to work correctly with the ESP32's SPI hardware?

Thanks
Ben

pierremolinaro, 2018-11-27:
Hi Ben,

Thank you for your message, it's nice to have thanks.

> I could not make it send or receive with traditional 500k CAN. Perhaps something wrong with oscillator settings or my use of a 20MHz resonator and not a true 40MHz crystal?
I did not try a 20MHz resonator, but I think it will work. 

I didn't use my driver with ESP32, I did not known that mSPI.usingInterrupt is not implemented.

On arduino, this function enables mSPI.beginTransaction to mask interrupts when called in user code. So only removing this line makes the driver buggy.

The workaround is to manually mask interrupt when needed.

Fortunatly (for us), the Teensy software has a bug, that requires to manually mask interrupt.

So take a look in ACAN2517.cpp to the tryToSend method: just remove conditional compilation for allways mask and unmask interrupts :
bool ACAN2517::tryToSend (const CANMessage & inMessage) {
  noInterrupts () ;
    mSPI.beginTransaction (mSPISettings) ;
      bool result = false ;
      if (inMessage.idx == 0) {
        result = enterInTransmitBuffer (inMessage) ;
      }else if (inMessage.idx == 255) {
        result = sendViaTXQ (inMessage) ;
      }
    mSPI.endTransaction () ;
  interrupts () ;
  return result ;
}

I hope that will solve the issue. If it works for the ACAN2517.cpp, il will work for the ACAN2515.cpp.

Best regards,

Pierre

pierremolinaro, 2019-01-22:
It appears that Arduino ESP32 interrupts are managed in a completely different way from usual Arduino:
- SPI.usingInterrupt is not implemented;
- noInterrupts() and interrupts() are NOPs;
- interrupt service routines should be fast, otherwise you get an "Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)".

So 1.0.x releases do not run on ESP32.

For a few days now, I have had an ESP32 connected to a MCP2517FD, so I will port the driver to this platform.

closed 2019-01-22 by pierremolinaro

Piter2004, 2019-10-14:
Can you help me with setting bautrate registers for sending at speed 33,333kb for opel.

What values i have to write to registers 
BRP
SEG1
SEG2
SJW
Thanks

pierremolinaro, 2019-10-14:
Hello,

You do need to write theses registers directly, the library computes the values from your speed (33,333kb) and the frequency  of the clock that drives the MCP2517FD.

Just open the LoopBackDemoESP32.ino and set :
	- the first argument according the clock (here ACAN2517Settings::OSC_4MHz10xPLL),
	- the second argument to your speed.

  ACAN2517Settings settings (ACAN2517Settings::OSC_4MHz10xPLL, 33333) ;
  settings.mRequestedMode = ACAN2517Settings::InternalLoopBack ; // Select loopback mode
  const uint32_t errorCode = can.begin (settings, [] { can.isr () ; }) ;
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Phase segment 1: ") ;
    Serial.println (settings.mPhaseSegment1) ;
    Serial.print ("Phase segment 2: ") ;
    Serial.println (settings.mPhaseSegment2) ;
    Serial.print ("SJW:") ;
    Serial.println (settings.mSJW) ;
    Serial.print ("Actual bit rate: ") ;
    Serial.print (settings.actualBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact bit rate ? ") ;
    Serial.println (settings.exactBitRate () ? "yes" : "no") ;
    Serial.print ("Sample point: ") ;
    Serial.print (settings.samplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }else{
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }

The serial prints that follow shows in the monitor the computed settings :
Bit Rate prescaler: 4
Phase segment 1: 239
Phase segment 2: 60
SJW:60
Actual bit rate: 33333 bit/s
Exact bit rate ? no
Sample point: 80%

No error, although the actual speed is slightly different. So : BRP = 4, PS1=239, …

Note, if you want to directly the registers, you should write theses values minus one.

Regards,

Pierre Molinaro