pierremolinaro/acan2517 #7: Arduino Uno with MCP2517FD

Neelakar544, 2019-11-06:
Hello All,

Thank you for the ACAN2517FD library provided for Arduino uno. I have tried Arduino uno loop-back demo code and configured all Arduino SPI lines to MCP2517FD click module,but below mentioned failure is setting up every time that i found in Arduino Serial Monitor. 
Please help me if anyone knows. It would great appreciation if you help others.  

Configure ACAN2517FD
MCP2517FD RAM Usage: 2016 bytes
Configuration error 0x1
Send failure

[image]

Best regards,
Neelakar Babu.

pierremolinaro, 2019-11-25:
Hello,

The 0x1 error is raised when the MCP2517FD does not reach the configuration mode in 2ms. It means that the MCP2517FD cannot be accessed via SPI, check the hardware connections.

Best Regards,

Pierre

Altantsooj, 2019-12-14:
Hello,
Same error here. 
I checked my connection. But I have the same error.

aileenfun, 2020-07-13:
hello,
anyone solved this problem?
I got 0x2. no idea why...