pierremolinaro/acan2517 #13: NRF52840 ACAN2518 at Normal20B receiving but not transmitting

pedrodpj, 2021-04-01:
Hi.
I'm using NRF52840 and library ACAN2717 version 1.1.10 and I'm facing some problem when trying to transmit frame data.
When running on loopback mode is ok, I can send and received frame data, but when running on Normal20B I can receive can data but when I transmit I get no error, but the data is not sent.

Se my code below:

    ACAN2517Settings settings (ACAN2517Settings::OSC_20MHz, 500 * 1000) ;
    settings.mRequestedMode = ACAN2517Settings::Normal20B;//InternalLoopBack;//

    //----------------------------------- Append filters
    ACAN2517Filters filters ;
    filters.appendFrameFilter (kStandard, PID_VIN, receiveVinData);
    filters.appendFrameFilter (kStandard, PID_KC_ODOMETER, receiveOdometerData);
    filters.appendFrameFilter (kStandard, PID_KC_SPEED, receiveSpeedData);
    filters.appendFrameFilter (kStandard, 0x777, receiveGeneric);
    filters.appendFrameFilter (kStandard, 0x778, receiveGeneric);
    filters.appendFrameFilter (kStandard, 0x779, receiveGeneric);

    //----------------------------------- Filters ok ?
    if (filters.filterStatus () != ACAN2517Filters::kFiltersOk) {
        debug.println("Error filter:  %d : %d ", filters.filterErrorIndex(), filters.filterStatus());
    }
    //----------------------------------- Enter configuration
    const uint32_t errorCode = can.begin (settings, [] { can.isr_core () ; }, filters) ;
    //----------------------------------- Config ok ?
    if (errorCode == 0)
    {
        debug.println("Bit Rate prescaler: %d", settings.mBitRatePrescaler);
        debug.println("Phase segment 1: %d", settings.mPhaseSegment1);
        debug.println("Phase segment 2: %d", settings.mPhaseSegment2);
        debug.println("SJW: %d", settings.mSJW);
        debug.println("Actual bit rate: %d bit/s", settings.actualBitRate());

        debug.println("Exact bit rate ? %s", settings.exactBitRate() ? "yes" : "no");
        debug.println("Sample point: %d%", settings.samplePointFromBitStart());
        debug.println("currentOperationMode: %d", can.currentOperationMode());
    }
    else
    {
        debug.println("Configuration error %x", errorCode);
    }

And now the log information:
Configure ACAN2517FD
Bit Rate prescaler: 1
Phase segment 1: 31
Phase segment 2: 8
SJW: 8
Actual bit rate: 500000 bit/s
Exact bit rate ? yes
Sample point: 80
currentOperationMode: 6
Configure ACAN2517FD done

At send operation I get:
Sent OK, with no error

Do you know if some addition settings is required to work with CAN20B at 500kbps?

pierremolinaro, 2021-04-02:
Hello,

There is no additionnal setting for sending at 500 kbit/s.

Do you have a logic analyzer, for checking the frame is not sent ?

Can you post your code, to see how you send message ?

What is the value of message.idx on sending ? Following figure 2 and §13 in the doc, the idx field value should be 0 or 255. It is ilmportant because if you reply to a received message, this field contains the index of the acceptance filter.

Pierre Molinaro

pedrodpj, 2021-04-05:
Hi Pierre,
Thanks for your quickly reply.
After check the schematics I've verified that the problem is related with transceiver TJA1057,The pin 8 was set to 3.3Vdc and this will force the transceiver to enter on "Silent Mode".
7.1.2 Silent modeA HIGH level on pin S selects Silent mode. The transmitter is disabled in Silent mode,releasing the bus pins to recessive state. All other IC functions, including the receiver,continue to operate as in Normal mode.

To solve I've fixed the hardware and set this pin to GND and now the library is working fine.

Thanks and best regards,Pedro

pierremolinaro, 2021-04-05:
Hi Pedro,

Ok, I understand, I am happy you have solved the problem.

Best Regards,

Pierre