pierremolinaro/acan2517 #16: My car Nexa has only implemented J1979 so no passive reading of CAN data. How to send and receive response

Som4567, 2022-01-06:
Hi 
My car Nexa has only implemented J1979 so I am not able to read raw CAN data using MCP2515 and also with MCP2517FD but with MCP2515 I am able to send requests to CAN like RPM and able to receive responses.  

My question is using MCP2517 how can I send the request and receive response like below code for MCp2515:

`if (useStandardAddressing) {
    CAN.beginPacket(0x7df, 8);
  } else {
    CAN.beginExtendedPacket(0x18db33f1, 8);
  }
  CAN.write(0x02); // number of additional bytes
  CAN.write(0x01); // show current data
  CAN.write(0x0c); // engine RPM
  CAN.endPacket();

  // wait for response
  while (CAN.parsePacket() == 0 ||
         CAN.read() < 3 ||          // correct length
         CAN.read() != 0x41 ||      // correct mode
         CAN.read() != 0x0c);       // correct PID

  float rpm = ((CAN.read() * 256.0) + CAN.read()) / 4.0;

  Serial.print("Engine RPM = ");
  Serial.println(rpm);`

Som4567, 2022-01-06:
`if(){}`

pierremolinaro, 2022-01-09:
Hello,

My librairies work only under Arduino, ACAN2515 with MCP2515, and ACAN2517 and ACAN2517FD with the MCP2517FD / MCP2518FD.

What is your system ? If you use an other system and / or other library, I can't help you.

However, the MC2517FD is not compatible with the MCP2515. If your code works with the MCP2515, it will not work with MCP2517FD, you should also change the library.

Pierre