pierremolinaro/acan2517 #20: Usage at MCP251863

Mirki, 2023-02-26:
Does anyone use this lib with MCP251863, what is actually a MCP2518 + Transceiver.
I'am not able to send messages in Normal20B

```
  ACAN2517Settings settings (ACAN2517Settings::OSC_40MHz, 125 * 1000) ; 
  settings.mRequestedMode = ACAN2517Settings::Normal20B ; 
  const uint32_t errorCode = can.begin (settings, NULL) ; 
```

But the SPI Connection ist working.

mickeyl, 2024-04-28:
@Mirki Were you able to solve your problem? I once had a MCP251863 as well, but couldn't find how to make it work with any driver code that worked with other boards due to some broken stand-by logic that was unique to my evaluation board. In theory, the MCP251863 should behave 100% identical to the 2518fd.

oxygenesaves, 2024-07-04:
u have any updates on this ?

mickeyl, 2024-07-04:
I had the same issue with a MCP25183 evaluation board from Mikroe – it was either defect or had a problem with the standby logic. A waveshare board worked instantly.