pierremolinaro/acan2517 #23: RaspberryPi Pico(RP2040) with MCP2518FDClick

rn-ksheer, 2024-06-06:
Hi Pierre,

I was successful in using acan2517 lib on Arduino nano with MCP2518FDClick board as sender and UNO with MCP2515 as receiver. I am unable to do the same with pico development board, The program compiles and uploads without any errors but not receiving any frames at UNO side. I have read your pdf document and made the changes to use custom SPI pins, still no success. Please point the necessary settings required to make pico work with MCP2518FDClick.

`SPI.setRX(16);
  SPI.setTX(19);
  SPI.setSCK(18);
  SPI.setCS(17);
  SPI.begin();

ACAN2517Settings settings (ACAN2517Settings::OSC_20MHz, 500UL * 1000UL) ;
  settings.mRequestedMode = ACAN2517Settings::Normal20B; // Select loopback mode`

Koryphon, 2024-06-06:
Hello @rn-ksheer 

These pins correspond to SPI1, not SPI (aka SPI0 according to the documentation)

Best regards

rn-ksheer, 2024-06-07:
hi @Koryphon ,

The pins I have used are default SPI (SPI0),
 Ref- [https://github.com/earlephilhower/arduino-pico/blob/master/variants/generic/pins_arduino.h#L18-L21](url)  &
[https://cdn-learn.adafruit.com/assets/assets/000/099/339/original/raspberry_pi_Pico-R3-Pinout-narrow.png?1612915004](url).

When i disconnect any one of the SPI pins from pico , i get **Configuration error 0x1**  & when all pins are connected the below info gets printed on serial monitor which confirms lib is communication with pico. 
```
sizeof (ACAN2517Settings): 32 bytes
Configure ACAN2517
MCP2517FD RAM Usage: 1024 bytes
Bit Rate prescaler: 1
Phase segment 1: 31
Phase segment 2: 8
SJW:8
Actual bit rate: 500000 bit/s
Exact bit rate ? yes
Sample point: 80%
Sent: 1
Sent: 2
```
I must be missing some other setting.
In ACAN2517.cpp, line 265 has _mSPISettings = SPISettings (800UL * 1000, MSBFIRST, SPI_MODE0) ;_ 
Does pico need clock speed adjustment here?

Koryphon, 2024-06-07:
Oups. Sorry. I checked the PIN number instead of the GPIO number.

rn-ksheer, 2024-06-11:
Finally got it working !!
Since i used default SPI pins, defining pin numbers in void setup was not required, SPI.begin handels it.
Only if alternate pins are being used it has to be SET in void setup.
_VIO SEL is set to 3v3 , STBY & STBY SEL are in their default position._ 
Tested with both 20MHz & 40MHz.

regards,
𝖱𝖾𝗇𝗎 𝖪𝗌𝗁𝗂𝗋𝗌𝖺𝗀𝖺𝗋

<head>
<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:\Users\user\AppData\Local\Temp\msohtmlclip1\01\clip.htm">
<link rel=File-List
href="file:///C:\Users\user\AppData\Local\Temp\msohtmlclip1\01\clip_filelist.xml">

</head>
<body link="#0563C1" vlink="#954F72">

pico | mcp2518FD
-- | --
GP16 | SDO
GP17 | CS
GP18 | SCK
GP19 | SDI
GP20 | INT
3V3 | 3V3
VBUS | 5V
GND | GND
</body>
</html>

[image]